jingpang/LearnVIORB #2: Test:: Scale initialize part work fine for my system

tdnet12434, 2017-01-11:
First of all, thank you for coding, Great work!! 

Yesterday i try to use your code in part of VIN init scale. It work well for my system (MPU6050+sj4000wifi). Even in large scale, the estimator still following environment scale as show in my [vid](https://www.youtube.com/watch?v=o-RZJxdjhXw&t=5) . By reducing the buffer of image callback, i got real-time estimate (this made lower fps). 

However, when i used full IMU tracking and IMU error for optimization. It seem still have scale misalignment at the return point like this [vid](https://youtu.be/bokmduLlBqs?t=122). I don't sure it cause by my camera time is not syncronized to IMU due to exposure time varied, or just normal limitation for monocular system. Have you try this case yet? 

Cheer.

jingpang, 2017-01-11:
Thanks for the discussion. 

I haven't tested such long-term/long-trajectory case. 
I think this may be an limitation of monocular system, because under degenerate motion the scale is not observable (e.g. const-velocity movement without sufficient acceleration excitation, see reference [6,14] in Raul's paper).
In such cases the system is equivalent to monocular vision-only system, so the scale-drift problem caused by accumulated error can't be avoided, especially with low-cost IMUs such as MPU6050. (However if the map is reused as in your first video, the scale can be consistent to the existing map).

This situation can be improved (not avoided) by global optimization in Loop Closure, or better sensor configuration (better IMU, good camera-IMU synchronization and good delay calibration with Kalibr, which reducing accumulated error).

----------------
[6] E. S. Jones and S. Soatto, “Visual-inertial navigation, mapping and localization: A scalable real-time causal approach,” The International Journal of Robotics Research, vol. 30, no. 4, pp. 407–430, 2011.
[14] Z. Yang and S. Shen, “Monocular visual-inertial state estimation with online initialization and camera-IMU extrinsic calibration,” IEEE Transactions on Automation Science and Engineering, 2016, DOI: 10.1109/TASE.2016.2550621 (to appear).

HeatherHu, 2017-03-30:
Thanks very much for your sharing. 
I tried the initialization part only without IMU+Vision Optimization on Euroc dataset(MH_01_easy.bag). What I did is use your code LocalMapping::TryInitVIO to get initial scale. After first try, the scale changes a lot in 15s.  But the final scale is far away from the real measurement scale. I see your test the code on this dataset. I am not sure if you tested the initial scale. 

Thanks

closed 2017-07-29 by jingpang