jingpang/LearnVIORB #3: regarding imu bias during preintegration

daobilige-su, 2017-01-21:
Hi,

1st, thanks for the awesome work.

just wandering in file IMUPreintegrator.cpp, on lines 108-110, the computation of _delta_P, _delta_V and _delta_R seems ignoring the IMU bias terms, which is present according the Eq.35-37 in the IMU preintegration paper. Is it because you are initialising the imu bias to be 0 or this bias is considered in somewhere else in the code? Thanks for your answer in advance. 

Have a nice weekend.

Cheers,

Su

jingpang, 2017-01-21:
Yes it is considered when calling the update(), e.g. in [Frame.cc](https://github.com/jingpang/LearnVIORB/blob/master/src/Frame.cc#L55).
The bias is subtracted here.

daobilige-su, 2017-01-21:
Thanks so much for your prompt reply. got it now.

Cheers,

Su

closed 2017-01-21 by daobilige-su