jingpang/LearnVIORB #4: Real time?

antithing, 2017-03-20:
Hi, thanks for making this code available! In the comments, you say 'not real-time', what is it that is stopping this from being real-time capable? 

Thanks!

jingpang, 2017-03-23:
Hi. Sorry i didn't notice the issue.

Tracking in current implementation will wait for LocalMapping, so it's not real-time.

To make it real-time, some small modifications are required:
1. Change the 'wait' in the data interface ( in ros_vio.cc/main(), wait SLAM.bLocalMapAcceptKF() )
2. Create a thread to do visual-inertial initialization ( it's slow and in Localmapping now )
3. Modify the NeedNewKeyFrame(), don't insert new keyframe if localmapping is not idle as in primary ORB_SLAM ( new keyframe will be inserted once 0.5s passed since last keyframe now )
4. some other small changes

antithing, 2017-03-23:
Thank you! I will try to implement these changes. Are you planning to release your real time capable version as well?  Thank you again for making this code available.

jingpang, 2017-03-24:
It still needs some debugging and I'm busy on some other things for the moment.

closed 2017-05-29 by jingpang