jingpang/LearnVIORB #5: Does it work in Windows

xuxu123, 2017-03-23:
Hello, Thanks for your work.
I am trying to compile it with noROS(https://github.com/OpenSLAM/LearnViORB_NOROS/tree/master/master/LearnVIORB_NOROS) in Windows, but there seems something error in the g2o.

If I complie the g2o in the Thirdparty, and use the g2o.lib to complie the ORB_SLAM2, it would be error during LINKing like this:
.......
2>g2otypes.obj : error LNK2001: 无法解析的外部符号 "public: virtual bool __cdecl g2o::OptimizableGraph::Edge::setMeasurementFromState(void)" (?setMeasurementFromState@Edge@OptimizableGraph@g2o@@UEAA_NXZ)
2>Optimizer.obj : error LNK2019: 无法解析的外部符号 "public: void __cdecl g2o::OptimizableGraph::Edge::setRobustKernel(class g2o::RobustKernel *)" (?setRobustKernel@Edge@OptimizableGraph@g2o@@QEAAXPEAVRobustKernel@3@@Z)，该符号在函数 "public: static void __cdecl ORB_SLAM2::Optimizer::BundleAdjustment(class std::vector<class ORB_SLAM2::KeyFrame *,class std::allocator<class ORB_SLAM2::KeyFrame *> > const &,class std::vector<class ORB_SLAM2::MapPoint *,class std::allocator<class ORB_SLAM2::MapPoint *> > const &,int,bool *,unsigned long,bool)" (?BundleAdjustment@Optimizer@ORB_SLAM2@@SAXAEBV?$vector@PEAVKeyFrame@ORB_SLAM2@@V?$allocator@PEAVKeyFrame@ORB_SLAM2@@@std@@@std@@AEBV?$vector@PEAVMapPoint@ORB_SLAM2@@V?$allocator@PEAVMapPoint@ORB_SLAM2@@@std@@@4@HPEA_NK_N@Z) 中被引用
......

I think there are something wrong in the linear_solver_cholmod.h, because I can't find any reference of CholmodExt and LinearSolverCholmod in the g2o. So I add a file to instance a LinearSolverCholmod and recompile it. But it doesn't work.

2>e:\vs2015project\viorb\learnviorb_noros\thirdparty\g2o\g2o\solvers\linear_solver_cholmod.h(207): error C2065: “MatrixXD”: 未声明的标识符
2> e:\vs2015project\viorb\learnviorb_noros\thirdparty\g2o\g2o\solvers\linear_solver_cholmod.h(375): note: 参见对正在编译的类 模板 实例化“g2o::LinearSolverCholmod”的引用
2>e:\vs2015project\viorb\learnviorb_noros\thirdparty\g2o\g2o\solvers\linear_solver_cholmod.h(207): error C2923: “g2o::SparseBlockMatrix”: 对于参数“MatrixType”，“MatrixXD”不是有效的 模板 类型变量
2>e:\vs2015project\viorb\learnviorb_noros\thirdparty\g2o\g2o\solvers\linear_solver_cholmod.h(268): error C3646: “_scalarPermutation”: 未知重写说明符
2>e:\vs2015project\viorb\learnviorb_noros\thirdparty\g2o\g2o\solvers\linear_solver_cholmod.h(268): error C2059: 语法错误:“,”
2>e:\vs2015project\viorb\learnviorb_noros\thirdparty\g2o\g2o\solvers\linear_solver_cholmod.h(268): error C2238: 意外的标记位于“;”之前
.......

Any advise to fix it ?

xuxu123, 2017-03-24:
I modify something in g2o (There are something different in the eigen), and recomplie the ORB_SLAM2. It is still two LNK error as follow:

错误	LNK2019	无法解析的外部符号 "bool __cdecl g2o::writeCCSMatrix(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int,int,int const *,int const *,double const *,bool)" (?writeCCSMatrix@g2o@@YA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@HHPEBH1PEBN_N@Z)，该符号在函数 "public: virtual bool __cdecl g2o::LinearSolverCholmod<class Eigen::Matrix<double,6,6,0,6,6> >::saveMatrix(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?saveMatrix@?$LinearSolverCholmod@V?$Matrix@N$05$05$0A@$05$05@Eigen@@@g2o@@UEAA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) 中被引用	ORB_SLAM2	E:\VS2015Project\VIORB\LearnVIORB_NOROS\build_c0x\Optimizer.obj	1	
错误	LNK2019	无法解析的外部符号 "public: __cdecl g2o::OptimizationAlgorithmGaussNewton::OptimizationAlgorithmGaussNewton(class g2o::Solver *)" (??0OptimizationAlgorithmGaussNewton@g2o@@QEAA@PEAVSolver@1@@Z)，该符号在函数 "public: static class Eigen::Matrix<double,3,1,0,3,1> __cdecl ORB_SLAM2::Optimizer::OptimizeInitialGyroBias(class std::vector<class ORB_SLAM2::KeyFrame *,class std::allocator<class ORB_SLAM2::KeyFrame *> > const &)" (?OptimizeInitialGyroBias@Optimizer@ORB_SLAM2@@SA?AV?$Matrix@N$02$00$0A@$02$00@Eigen@@AEBV?$vector@PEAVKeyFrame@ORB_SLAM2@@V?$allocator@PEAVKeyFrame@ORB_SLAM2@@@std@@@std@@@Z) 中被引用	ORB_SLAM2	E:\VS2015Project\VIORB\LearnVIORB_NOROS\build_c0x\Optimizer.obj	1	

Any advise ?

jingpang, 2017-03-24:
I only tested this project on Ubuntu 14.04. I've no idea about the compile error on windows...

closed 2017-05-29 by jingpang