jingpang/LearnVIORB #6: Shows segmentation fault while trying to run on Ubuntu 16.04

arin07, 2017-03-30:
Hi
While trying to run the code, I run into SIGSEGV error (probably segmentation fault) and the program exits. I tried debugging in qtcreator and found that it runs into the above problem at  line 160 in ros_vio.cc  "SLAM.TrackMonoVI(im, vimuData, imageMsg->header.stamp.toSec() - imageMsgDelaySec);" . I am new to this and have not been yet able to figure out what is causing this. Any help would be appreciated .

jingpang, 2017-04-01:
I just tested on ubuntu14.04... You may need to debug it..

imrasp, 2017-06-29:
I am using 16.04 and running into core dumped like you but it's Aborted! not segfault. the result of gdb is shown below. Do you know how to solve this?  

`... Map scale updated ...

... Map NavState updated ...

LearnVI_Drone: /home/rasp/CLionProjects/LearnVI_Drone/VIORB/src/IMU/so3.cpp:122: Sophus::SO3::SO3(const Quaterniond&): Assertion `unit_quaternion_.squaredNorm() > SMALL_EPS' failed.

Thread 5 "LearnVI_Drone" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffc7462700 (LWP 3011)]
0x00007ffff08fd428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff08fd428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff08ff02a in __GI_abort () at abort.c:89
#2  0x00007ffff08f5bd7 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0xd0dd20 "unit_quaternion_.squaredNorm() > SMALL_EPS", 
    file=file@entry=0xd0dce0 "/home/rasp/CLionProjects/LearnVI_Drone/VIORB/src/IMU/so3.cpp", line=line@entry=122, 
    function=function@entry=0xd10f00 <Sophus::SO3::SO3(Eigen::Quaternion<double, 0> const&)::__PRETTY_FUNCTION__> "Sophus::SO3::SO3(const Quaterniond&)") at assert.c:92
#3  0x00007ffff08f5c82 in __GI___assert_fail (assertion=0xd0dd20 "unit_quaternion_.squaredNorm() > SMALL_EPS", file=0xd0dce0 "/home/rasp/CLionProjects/LearnVI_Drone/VIORB/src/IMU/so3.cpp", line=122, 
    function=0xd10f00 <Sophus::SO3::SO3(Eigen::Quaternion<double, 0> const&)::__PRETTY_FUNCTION__> "Sophus::SO3::SO3(const Quaterniond&)") at assert.c:101
#4  0x00000000006ed959 in Sophus::SO3::SO3(Eigen::Quaternion<double, 0> const&) ()
#5  0x00000000006ee27a in Sophus::SO3::expAndTheta(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, double*) ()
#6  0x00000000006ee090 in Sophus::SO3::exp(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) ()
#7  0x00000000006f193b in ORB_SLAM2::NavState::IncSmallPR(Eigen::Matrix<double, 6, 1, 0, 6, 1>) ()
#8  0x000000000053385b in g2o::VertexNavStatePR::oplusImpl(double const*) ()
#9  0x00007ffff6a4c117 in g2o::SparseOptimizer::update(double const*) () from /home/rasp/CLionProjects/LearnVI_Drone/VIORB/Thirdparty/g2o/lib/libg2o.so
#10 0x00007ffff6a5659a in g2o::OptimizationAlgorithmLevenberg::solve(int, bool) () from /home/rasp/CLionProjects/LearnVI_Drone/VIORB/Thirdparty/g2o/lib/libg2o.so
#11 0x00007ffff6a4d937 in g2o::SparseOptimizer::optimize(int, bool) () from /home/rasp/CLionProjects/LearnVI_Drone/VIORB/Thirdparty/g2o/lib/libg2o.so
#12 0x000000000051495e in ORB_SLAM2::Optimizer::GlobalBundleAdjustmentNavStatePRV(ORB_SLAM2::Map*, cv::Mat const&, int, bool*, unsigned long, bool) ()
#13 0x00000000004a9b84 in ORB_SLAM2::LocalMapping::TryInitVIO() ()
#14 0x00000000004acbd6 in ORB_SLAM2::LocalMapping::Run() ()
#15 0x000000000047092d in void std::_Mem_fn_base<void (ORB_SLAM2::LocalMapping::*)(), true>::operator()<, void>(ORB_SLAM2::LocalMapping*) const ()
#16 0x000000000046ec23 in void std::_Bind_simple<std::_Mem_fn<void (ORB_SLAM2::LocalMapping::*)()> (ORB_SLAM2::LocalMapping*)>::_M_invoke<0ul>(std::_Index_tuple<0ul>) ()
#17 0x000000000046c600 in std::_Bind_simple<std::_Mem_fn<void (ORB_SLAM2::LocalMapping::*)()> (ORB_SLAM2::LocalMapping*)>::operator()() ()
#18 0x000000000046a8a4 in std::thread::_Impl<std::_Bind_simple<std::_Mem_fn<void (ORB_SLAM2::LocalMapping::*)()> (ORB_SLAM2::LocalMapping*)> >::_M_run() ()
#19 0x00007ffff1fbac80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#20 0x00007ffff7bc16ba in start_thread (arg=0x7fffc7462700) at pthread_create.c:333
#21 0x00007ffff09cf3dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
`

Thank you in advance :)

hitleeleo, 2017-12-27:
@imrasp Did you solve this issue?

NCHUPJ, 2019-09-23:
@imrasp @hitleeleo Did you solve this issue?

iliferobot, 2019-10-22:
@imrasp Hi, i encountered the same problem as yours, it shows segfault on so3, i debug it with gdb but still has no idea where this bug from . 
Do yo have any idea or have you solved this problem?

ns15417, 2019-10-26:
@NCHUPJ  Hi, dont know whether you fixed this problem and I already fixed it by add EIGEN_MAKE_ALIGNED_OPERATOR_NEW inside the class and also change the type of mv20FramesReloc from vector<Frame> to vector<Frame,Eigen::aligned_allocator<Frame>>
[image]
you can have a try to see if it works

yanmengting117, 2019-11-14:
i encountered the same problem when running the KITTI dataset.
Do you know how to deal with it?

ns15417, 2019-12-03:
@yanmengting117  as i said in the last comment, you need to add EIGEN_MAKE_ALIGNED_OPERATOR_NEW to the file which i forgot which file but it is the file containing the variable mv20FramesReloc;
when you find the file, first, add EIGEN_MAKE_ALIGNED_OPERATOR_NEW inside the class, and second, change mv20FramesReloc from  vector to vector<Frame,Eigen::aligned_allocator>