jingpang/LearnVIORB #7: Is this algorithm be tested?

gogojjh, 2017-04-01:
Too awesome.
I wonder if this algorithm was compared with the groundtruth in Euroc, KITTI or TUM RGB-D dataset? 
Thank you!

jingpang, 2017-04-01:
I tested in Euroc and compare the KeyFrame's trajectories with ground truth , the RMSEs are larger than those in the paper. 

For example the RMSE of V101_easy is about 7cm, and 12cm for MH01_easy. (Scale is not aligned for the above results. The scale is about 0.97~1.03 and I think the initialization has the major influence on scale error.)

Compare with the scripts by [raul](https://github.com/raulmur/evaluate_ate_scale) and by [TUM](http://vision.in.tum.de/data/datasets/rgbd-dataset/tools#evaluation), or use [this one](https://github.com/QingfengLi-hit/coordinate_align) to align the coordinate. You can have a try of them.

gogojjh, 2017-04-03:
And I would like to know which parameters or steps can be changed to speed up the algorithm? Could you give me some ideas?

jingpang, 2017-04-03:
Manage to insert less KeyFrames (change the strategies of NeedNewKeyFrame in Tracking and KeyFrameCulling in LocalMapping), and perhaps better coding.
And extracting less keypoints may speed up (but the precision is worse, need balance).

closed 2017-05-29 by jingpang