jingpang/LearnVIORB #8: Test Initial Scale?

HeatherHu, 2017-04-05:
Really thanks for sharing your code. I tried the initial part, but I didn't get the correct scale. What I did is using your code TryInitVIO after the number of key frames is larger than 10. Considering the error accumulation of IMU, every image frame is consider as key frame. I am not sure if I ignore other steps. 
I'd like to know if you test the initial scale on Euroc dataset. Thanks.

jingpang, 2017-04-05:
I tested the initialization with EuRoc dataset, and the estimated scale is about 0.9x ~ 1.0x。

The reason why your result is bad may be lack of KeyFrames. According to Raul's paper, to achieve good initialization result in EuRoc, about 15 seconds is needed. There'll be ~100 KeyFrames. 

In fact, to achieve good initialization result ( initialization is a linear estimation procedure assuming fixed IMU bias ), the motion of sensor is required to fulfill the observability property of visual-inertial system, i.e. enough acceleration stimulus on all 3-axis. So perhaps using only 10 frames is not enough.

About observability, you can refer to Raul's paper and the reference paper.

HeatherHu, 2017-04-06:
Thanks for your quick reply. In your code I found every time your reset all variables(bg,ba) to zero. It means the last estimated value is not used for the new estimation. If this it seems that only the last initialization result is useful. Even if you already set bg and ba as zero, why you do imu integration every time when calling TryInitiVIO. I am not sure if I misunderstanding.

jingpang, 2017-04-07:
Yes, only the last computation's result is used. I computed all the results in the 15s to check the change, as the Fig.4 shows in Raul's paper.

The bg is set and preintegration is recomputed during initialization, so after reset the integration also need recomputation.

HeatherHu, 2017-04-10:
I just can't understand why the initialization is computed many times. Like you said all stuff is recomputed in one initialization step without using the last result. Why these variables tend to stable after sometime. Is this just because of the increase of key frames? Thanks.

jingpang, 2017-07-29:
Sorry I didn't note this comment..
The answer is the same as above on Apr.5. The system need sufficient acceleration & rotation stimuli to estimate the scale/bias accurately. 
The theory about this problem is observability, you can google 'visual inertial observability' and find some papers about this.

closed 2017-07-29 by jingpang