jingpang/LearnVIORB #11: Unstable accelerometer bias estimation

Mikhail2017, 2017-04-27:
Hi.
I have tested your implementation on EurocMAV dataset.
On all my tests I've got robust gyroscope & gravity etsimitation.
But unfotunately accelerometer is wrong on some scenes (V1_01_easy for example).
Also it loooks like it heavely depends on initial scene scale. What's wrong?
Also is there any implmentation of orb slam with GTSAM library?
It looks like you have converted some GTSAM factors to orb library and to g2o optimization.

jingpang, 2017-05-02:
The ground truth values for accelerometer bias of V1_01_easy seem strange. I don't think the value can be as large as about 0.5m/s^2.  For other scenes that value is about 0.1~0.2 which is closer to the estimated value.

As the scale is coupled with acc-bias in the estimator, so the estimated values tend to be wrong/right together. If the scale&acc-bias error of this initialization method is too large, I think the main reason may be the motion is not enough and making the observability not so good.

For the last question, yes. I read the code of IMU preintegration factor in GTSAM and referred to the library. But I've no idea about any implementation of ORB-SLAM with GTSAM.

yan-lu, 2017-06-16:
Nice work.  Just curious, is there a way to verify whether the preintegration is computed correctly?

jingpang, 2017-06-16:
Thx. 

I don't know how to verify the results online... 

BTW, what do you mean by 'computed correctly'? As the preintegration computation only requires imu measurements and estimated bias, so the errors may come from wrong measurement (e.g. sensor shock) or wrongly estimated bias. Do you mean how to monitor or evaluate such error?

yan-lu, 2017-06-16:
I wonder why the order of PRV or PVR matters. Can you give me a reference? Thanks.

jingpang, 2017-06-17:
In fact no fundamental difference. I just change the order to split the PVR vertex to PR+V, as in the paper.

yan-lu, 2017-06-17:
By "the paper", do you mean "Visual-Inertial Monocular SLAM with Map Reuse" or "IMU Preintegration on Manifold .."?

jingpang, 2017-06-17:
"Visual-Inertial Monocular SLAM with Map Reuse"

closed 2017-07-29 by jingpang