jingpang/LearnVIORB #12: code problem

zdczdcc, 2017-05-31:
In IMUPreintegrator.h,the function `JacobianR` calculates the right jacobian by the formula 
`JR = I - (1- cos( theta ) ) / theta * K + ( 1 - sin( theta ) / theta ) * K * K`.But i saw the formula to do the same thing in Gao Xiang's blog was `JR = sin( theta ) / theta * I - (1- cos( theta ) ) / theta * K + ( 1 - sin( theta ) / theta ) * k * k.transpose()`.I wonder if i was wrong or misunderstood somewhere.

jingpang, 2017-05-31:
It's the same. As 
k*k.transpose = k^*k^ + (k.transpose*k)*I

Refer to 'SO(3) identities and approximations' in about pg244 of the book 'State estimation' by Barfoot.

closed 2017-05-31 by jingpang

zdczdcc, 2017-06-01:
Sorry,i haven't found the formula about `k * k.transpose = k^ * k^ + (k.transpose * k) *I` about the page range 231-246 of the book you mentioned.But with that formula,they are just the same.

jingpang, 2017-06-01:
The equation can be found [here](http://asrl.utias.utoronto.ca/~tdb/bib/barfoot_ser17_identities.pdf).
The 7th equation in the column 'Lie Algebra'.

zdczdcc, 2017-06-01:
I've found that with your help.Thanks a lot!

zdczdcc, 2017-06-07:
In function `IMUPreintegrator::update()`,how to calculate the variable `_cov_P_V_Phi`,`_J_(P/V/R)_Bias(g/a)`.I wonder to know the reference of them.And i found the reference of the variable `_delta_(P/V/R)`'s  on paper `On-Manifold Preintegration for Real-Time
Visual-Inertial Odometry`.But  the `_J_(P/V/R)_Bias(g/a)` has a little difference with the paper i mentioned.

jingpang, 2017-06-07:
It's the same as Appendix.B in the paper you mentioned.
But in the code it's implemented as J(i+1) = J(i) + J_add, which can be easily derived from the equation.

zdczdcc, 2017-06-08:
In the paper,the `_J_R_Biasg = dR.transpose()*_J_R_Biasg * Jr*dt`.But in your code,it was 
`_J_R_Biasg = dR.transpose()*_J_R_Biasg - Jr*dt`.

And i found the noise covariance propagation of delta measurements in the Appendix.E.But i couldn't correlate these part with your code about `_cov_P_V_Phi`.

xukuanHIT, 2017-12-09:
@zdczdcc 
Hi， I hope it can help you
[image]

zdczdcc, 2017-12-19:
@xukuanHIT ,
I've understood it with your help.Thanks a lot!

lianbin, 2018-10-16:
> @zdczdcc
> Hi， I hope it can help you
> [image]

  But ,in the code ,why compute delta Ri j+1 / deltabg   but  not  delta Ri j / deltabg as the paper?  Thank you.