jingpang/LearnVIORB #13: Building issue: "Could NOT find CHOLMOD"

marco-tranzatto, 2017-06-12:
Hi,

Thanks for sharing this repo!
I'm trying to test it, but when I execute the `build.sh` script I get the following error:
```
Could NOT find CHOLMOD (missing:  CHOLMOD_INCLUDE_DIR CHOLMOD_LIBRARIES) 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CHOLMOD_INCLUDE_DIR
   used as include directory in directory /home/marco/viorb_ws/src/LearnVIORB
CHOLMOD_LIBRARY
    linked by target "ORB_SLAM2" in directory /home/marco/viorb_ws/src/LearnVIORB

-- Configuring incomplete, errors occurred!
```

Any idea why?

Thanks,
Marco.

jingpang, 2017-06-13:
Maybe you need to install cholmod by 'sudo apt-get install libsuitesparse-dev'.

marco-tranzatto, 2017-06-13:
thanks! :+1:

closed 2017-06-13 by marco-tranzatto