jingpang/LearnVIORB #15: two questions on PoseOptimization

IQ17, 2017-06-20:
Hi, thank you very much for sharing the code!
I am learning your code recently, and I have two questions on PoseOptimization functions, could you give me some instructions? Thanks in advance!

1) how you determine the values for Huber? 
for example "const float thHuberNavState = sqrt(30.5779);", how this 3.5779 is found?

2) about marginalization, it seems you marginalized vertexs of the current frame, shouldn't we marginalize vertexs of the last frame/keyframe instead?

    // Compute marginalized Hessian H and B, H*x=B, H/B can be used as prior for next optimization in PoseOptimization

    if(bComputeMarg)

    {

        std::vector<g2o::OptimizableGraph::Vertex*> margVerteces;

        margVerteces.push_back(optimizer.vertex(FramePVRId));

        margVerteces.push_back(optimizer.vertex(FrameBiasId));

        g2o::SparseBlockMatrixXd spinv;

        optimizer.computeMarginals(spinv, margVerteces);

        // spinv include 2 blocks, 9x9-(0,0) for PVR, 6x6-(1,1) for Bias

        Matrix<double,15,15> margCov = Matrix<double,15,15>::Zero();

        margCov.topLeftCorner(9,9) = spinv.block(0,0)->eval();

        margCov.topRightCorner(9,6) = spinv.block(0,1)->eval();

        margCov.bottomLeftCorner(6,9) = spinv.block(1,0)->eval();

        margCov.bottomRightCorner(6,6) = spinv.block(1,1)->eval();

        pFrame->mMargCovInv = margCov.inverse();

        pFrame->mNavStatePrior = ns_recov;

        //Debug log

        //cout<<"inv MargCov 2: "<<endl<<pFrame->mMargCovInv<<endl;

    }

jingpang, 2017-06-21:
Hi.

1. To determine Huber threshold, i use _chi2inv()_ function in MATLAB/Octave. For example, chi2inv(0.95,2)=5.991 is the 2D threshold for image feature points.
You can try [Octave online](https://octave-online.net/) if you have not installed MATLAB/Octave.

2. As far as I know, the main idea of marginalization is the same as you mentioned, but in g2o the usage is like the code. You can find the discussion in [this issue of g2o](https://github.com/RainerKuemmerle/g2o/issues/83).

Jing

IQ17, 2017-06-22:
Hi, thank you for your instruction!

Today I tried to cout the obtained NavStatePrior, and I found that NavStatePrior.Get_P() is the same as the pFrame->GetNavState().Get_P(). That is, the NavStatePrior is the same as the current frame NavState. However, I was expected to see that NavStatePrior is more closer to the last frame/keyframe NavState....

Then I changed to in code as below, but g2o reports run-time errors and tracking gets lost
std::vector<g2o::OptimizableGraph::Vertex*> margVerteces;
 margVerteces.push_back(optimizer.vertex(LastFramePVRId)); // it was FramePVRId
 margVerteces.push_back(optimizer.vertex(LastFrameBiasId)); // it was FrameBiasId

When I experienced this problem, I am NOT using your github code directly. I work under window, and I am adding your code piece by piece to the VS solution, so maybe I missed something...

So my question actually is that shouldn't NavStatePrior be more closer to the last frame/keyframe NavState, than the current frame NavState?

thanks！

jingpang, 2017-06-23:
The purpose of marginalization is to compute an approximate marginal (Gaussian) distribution of current frame. The mean is the estimated state for current frame, and the same for covariance. So as in [the code](https://github.com/jingpang/LearnVIORB/blob/RT/src/Optimizer.cc#L2032) the mean and covariance are both for current frame. 

And this mean and covariance are stored in the 'current' and will be used when the 'current' becomes 'last' frame in the next optimization.

When you build the optimizer, the prior (mean and covariance) of 'last' frame is used [here in the code](https://github.com/jingpang/LearnVIORB/blob/RT/src/Optimizer.cc#L1746).

IQ17, 2017-06-29:
Hi, thanks for your instructions!
I just got VI-ORB work in my system, and it works well with Euroc dataset, thank you very much!

And I have a few of more questions...and they all related to the PoseOptimization
1) about `eNSBias->setInformation(InvCovBgaRW/imupreint.getDeltaTime());`, shouldn't we use `eNSBias->setInformation(InvCovBgaRW * imupreint.getDeltaTime());` instead? Because in the preintegration theory paper, it is Cov/dt, so I think for InvCov may be we should use InvCov*dt...

2) [Here](https://github.com/jingpang/LearnVIORB/blob/master/src/Optimizer.cc#L642) you updated the pFrame, but not the pLastFrame, why it is so? As I understand, pLastFrame's NavState is also adjusted and thus should be updated. Maybe you think that the pLastFrame has been outputed and is not a KeyFrame, but we actually will enter this PoseOptimization function twice and thus use pLastFrame twice. 

3) The optimizer always run 4 times, with 10 iterations each time, as shown [here](https://github.com/jingpang/LearnVIORB/blob/master/src/Optimizer.cc#L546). But I tried to print out the total error for all 40 iterations, and found that the total error got stable with at most 3 iterations. With 3 iterations, the trajectory ATE is also good. So maybe another 37 iterations can be saved to boost FPs.

jingpang, 2017-06-30:
Hi. Thanks for your discussion. My understanding for the questions above:

1. The discrete noise covariance is Cov/dt, but for bias, it's Cov*dt.
You can check it in [Forster's PhD thesis](http://rpg.ifi.uzh.ch/docs/PhD16_Forster.pdf). (Sec. 'Bias Model', the paragraph below Equation (C.47).) 

2. I just forgot and ignored that. You can have a try.

3. The stop criteria of Levenberg or Gauss-Newton method is added in the solver in g2o. For example Raul added the code [here](https://github.com/jingpang/LearnVIORB/blob/master/Thirdparty/g2o/g2o/core/optimization_algorithm_levenberg.cpp#L154). You may change the criteria.
Also you can add some criteria for Gauss-Newton as I did [here in real-time branch](https://github.com/jingpang/LearnVIORB/blob/RT/Thirdparty/g2o/g2o/core/optimization_algorithm_gauss_newton.cpp#L97).

closed 2017-07-29 by jingpang

XinnWang, 2021-12-31:
@IQ17 @jingpang Hi, i have a similar question about pLastFrame inside PoseOptimization. pLastFrame is already optimized with the frame before pLastFrame, why should pLastFrame be optimized again with current frame ?