jingpang/LearnVIORB #16: process has died when run offline dataset on ubuntu 16

HiZhongfei, 2017-07-03:
[VIO-2] process has died [pid 12821, exit code -11, cmd /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/VIO /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/../../../Vocabulary/ORBvoc.bin /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/../../../config/euroc.yaml __name:=VIO __log:=/home/luozhongfei/.ros/log/b0b00082-5fd3-11e7-bf6c-484d7ec87cb6/VIO-2.log].
log file: /home/luozhongfei/.ros/log/b0b00082-5fd3-11e7-bf6c-484d7ec87cb6/VIO-2*.log

jingpang, 2017-07-12:
Sorry it's only tested on Ubuntu 14.04...

mrteera, 2017-07-29:
I also have a problem with Ubuntu 16.04. If you host machine is Ubuntu 16.04, you can try to use Docker with Ubuntu 14.04 like this: https://github.com/jingpang/LearnVIORB/pull/26.

jingpang, 2017-07-29:
Thanks. @mrteera

guilhermelawless, 2017-07-29:
I've got it working on Ubuntu 16.04. You can try to apply #22 , but it's useful if you could run the process with gdb and backtrace it.

imrasp, 2017-07-30:
I used to have the same error as you before when the ros process start, but today I've got it working on Ubuntu 16.04 with ROS Kinetic. This is a part of my previous log file 

`[roslaunch][INFO] 2017-07-30 16:48:45,654: ... launch_nodes complete
[roslaunch.pmon][INFO] 2017-07-30 16:48:45,654: registrations completed <ProcessMonitor(ProcessMonitor-1, started daemon 140184457225984)>
[roslaunch.parent][INFO] 2017-07-30 16:48:45,654: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2017-07-30 16:48:45,654: spin
[roslaunch][ERROR] 2017-07-30 16:48:46,670: **[VIO-2] process has died [pid 6954, exit code -11**, cmd /home/rasp/CLionProjects/LearnVI_Drone/VIORB/Examples/ROS/ORB_VIO/VIO /home/rasp/CLionProjects/LearnVI_Drone/VIORB/Examples/ROS/ORB_VIO/../../../Vocabulary/ORBvoc.bin /home/rasp/CLionProjects/LearnVI_Drone/VIORB/config/euroc.yaml __name:=VIO __log:=/home/rasp/.ros/log/45a66774-750c-11e7-8023-3863bba9c370/VIO-2.log].
log file: /home/rasp/.ros/log/45a66774-750c-11e7-8023-3863bba9c370/VIO-2*.log
[roslaunch.pmon][INFO] 2017-07-30 16:48:46,671: ProcessMonitor.unregister[VIO-2] starting
[roslaunch.pmon][INFO] 2017-07-30 16:48:46,671: ProcessMonitor.unregister[VIO-2] complete
`

Follow @mrteera  docker file, I reinstall Eigen from the latest version to 3.2.10 as using in his docker [here](https://github.com/mrteera/LearnVIORB/blob/ad7fc9fcafa82311c49b01134c7a299b85a04201/Dockerfile#L137), install [cv_bridge](http://wiki.ros.org/cv_bridge) and ros opencv3 (`sudo apt-get install ros-kinetic-opencv3`) then it work! 

Hope this help :)

tdnet12434, 2017-07-31:
Hi, @imrasp  
I interest in your work. Which computer you are using if it running onboard to complete the SLAM part? and  How much performance it can achieved? 
Thank you advanced.

imrasp, 2017-08-09:
@tdnet12434  I use Odroid XU4 install Ubuntu 14.04 without GUI for SLAM part, it can achieve only (approx.) 3 fps after initialization without view.

DEARsunshine, 2019-10-15:
> [VIO-2] process has died [pid 12821, exit code -11, cmd /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/VIO /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/../../../Vocabulary/ORBvoc.bin /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/../../../config/euroc.yaml __name:=VIO __log:=/home/luozhongfei/.ros/log/b0b00082-5fd3-11e7-bf6c-484d7ec87cb6/VIO-2.log].
> log file: /home/luozhongfei/.ros/log/b0b00082-5fd3-11e7-bf6c-484d7ec87cb6/VIO-2*.logwould 

would you share how to solver it ?

CQnancy, 2023-07-31:
> > [VIO-2] process has died [pid 12821, exit code -11, cmd /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/VIO /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/../../../Vocabulary/ORBvoc.bin /home/luozhongfei/LearnVIORB/Examples/ROS/ORB_VIO/../../../config/euroc.yaml __name:=VIO __log:=/home/luozhongfei/.ros/log/b0b00082-5fd3-11e7-bf6c-484d7ec87cb6/VIO-2.log].
> > log file: /home/luozhongfei/.ros/log/b0b00082-5fd3-11e7-bf6c-484d7ec87cb6/VIO-2*.logwould
> 
> @luoru would you share how to solver it ?

@luoru Hello would you share how to solver it ?