jingpang/LearnVIORB #19: feature lost when after TryInitVIO return true

wangminhang, 2017-07-12:
I run this program in realtime with RealSense Camaro，when the function LocalMapping::TryInitVIO() returns true , all features lost and can't be catched. 
How to solve this issue? Thank you！

Also in this function，sometimes the scale and depth is negative，why？

jingpang, 2017-07-12:
Maybe the image and imu messages are not well synchronized, so the initialization may get wrong result.

closed 2017-07-29 by jingpang

Veerachart, 2017-08-18:
@wangminhang has you happened to get some solution to this?

I also got some similar problem, with scale values sometime positive sometime negative. If I am lucky, then positive value is obtained, but the tracking seems not working so well and I don't think the scale is correct. Most of the time I end up losing the track.