jingpang/LearnVIORB #23: How do  you evaluate the results of VI-ORB against the groundtruth on EuRoC dataset?

nkwangyh, 2017-07-24:
It seems that the timestamps and FPS of the groundtruth and the output trojectory of VI-ORB are not the same. What's more, the groundtruth trojectory and the VI-ORB trojectory are based on different global coordinate systems(There is a transformation(rotation and translation) between the groundtruth and VI-ORB). So how do you evaluate the results of VI-ORB against the groundtruth or original ORB?

closed 2017-07-25 by nkwangyh

guilhermelawless, 2017-07-25:
Why close? @nkwangyh

einspyon, 2017-11-19:
Had you solved the problem now ??? 
How??

Pratapabhay, 2018-01-16:
Had you solved it?

Markbess, 2018-01-18:
Any ideas? I just know that there are some evaluation tools in TUM dataset fot RGB-D data
https://vision.in.tum.de/data/datasets/rgbd-dataset/tools#evaluation

Pratapabhay, 2018-02-10:
@Markbess @jingpang 
Hi Markbess, can monocular vi-orb produce optimized trajectory(Full BA) for all the frames, not just key frames? I tried exploiting the code and was able to get non-optimized trajectory. In the paper **"Visual-Inertial Monocular SLAM with Map Reuse"** by Raul Mur-Artal, he did measured the performance of the system. Did he do it only on keyframes or is there a way to do it on all frames? Thanks in advance.

Markbess, 2018-02-10:
Yeah,I guess you can do it by adding all frames  to keyframes, but it is
costly to compute all frames.

Pratapabhay, 2018-02-10:
Thanks for the reply.
I am not able to figure out how to control the keyframe generation? Could you suggest how to do it? or is there a way to directly add all frames to keyframes?