jingpang/LearnVIORB #28: Camera-imu frame transformation What is the parameter Tic Pc?

zgq91, 2017-08-02:
Hi: 
    I can't understand what the Tic Pc parameter means.

RachitB11, 2017-08-15:
Pc is the point in the camera frame. Tic converts the point from the camera to the imu frame.
I don't know why the parameter names the parameter as Camera.Tbc in the yaml.

rocinant, 2017-08-17:
Tic means the pose matrix of ith camera frame, Pc  means the coordnator of map point in this camera frame. The name of Camera.Tbc means extrinsic transition matrix from body of IMU to camera from IMU-camera calibration. If you asked this question, then it means you mostly never read the related paper but directly put your hands on the code. The name conventions are mostly right. I will point you guys a correct direction, read the related paper of Raul Mur-Artal about ORB visual SLAM and Forster's papers about IMU preintegration, then read the code, otherwise you guys will keep asking this kind of questions.

imrasp, 2018-01-11:
[image]
@rocinant  refer to image from paper "On-Manifold Preintegration for Real-Time Visual--Inertial Odometry" of Forster, Is the Tbc is extrinsic transition matrix from camera to body of IMU?