jingpang/LearnVIORB #32: ./build.sh error

cumtchenchang, 2017-09-30:
cc@cc-K55VD:~/catkin_ws/src/LearnVIORB$ ./build.sh 
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cc/catkin_ws/src/LearnVIORB/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cc/catkin_ws/src/LearnVIORB/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Could NOT find CHOLMOD (missing:  CHOLMOD_INCLUDE_DIR CHOLMOD_LIBRARIES) 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CHOLMOD_INCLUDE_DIR
   used as include directory in directory /home/cc/catkin_ws/src/LearnVIORB
   used as include directory in directory /home/cc/catkin_ws/src/LearnVIORB
   used as include directory in directory /home/cc/catkin_ws/src/LearnVIORB
CHOLMOD_LIBRARY
    linked by target "ORB_SLAM2" in directory /home/cc/catkin_ws/src/LearnVIORB

-- Configuring incomplete, errors occurred!
See also "/home/cc/catkin_ws/src/LearnVIORB/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
Build ROS node ...
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_VIO
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using CATKIN_DEVEL_PREFIX: /home/cc/catkin_ws/src/LearnVIORB/Examples/ROS/ORB_VIO/build/devel
-- Using CMAKE_PREFIX_PATH: /home/cc/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/cc/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/cc/catkin_ws/src/LearnVIORB/Examples/ROS/ORB_VIO/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[rosbuild] Including /opt/ros/kinetic/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/kinetic/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/kinetic/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cc/catkin_ws/src/LearnVIORB/Examples/ROS/ORB_VIO/build
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../VIO'.  Stop.
CMakeFiles/Makefile2:163: recipe for target 'CMakeFiles/VIO.dir/all' failed
make[1]: *** [CMakeFiles/VIO.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Launch file in Examples/ROS/ORB_VIO/launch.
Modify the configuration file config/euroc.yaml
Run as: roslaunch ORB_VIO testeuroc.launch

jixiedaima, 2017-12-26:
Do you solve it, I met the same problem?

BoomFan, 2018-07-02:
Same:
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../VIO'. Stop.

Any solution yet?

connorsoohoo, 2018-07-30:
Bump. Got the same build error here.

lwcode, 2019-05-24:
same problem.
Have solution yet?

lwcode, 2019-05-24:
I have find the solution:
the real cause of problem is "Could NOT find CHOLMOD"
so ,first:
install libsuitesparse-dev , it will be ok.

otherwise:
maybe have the Eigen problem,see this: https://github.com/jingpang/LearnVIORB/issues/36
and open file "Examples/ROS/ORB_VIO/CMakeLists.txt",change line "find_package(OpenCV 2.4.3 REQUIRED)" to your own opencv version.