jingpang/LearnVIORB #35: Two questions about your code

einspyon, 2017-11-02:
Hi ! You really did a nice work, thanks so much !!!
There're two problems which trouble me,  beg your help !!!

1. About IMU noise model 

   Matrix3d IMUData::_gyrMeasCov = Matrix3d::Identity()*1.7e-4*1.7e-4/0.005/**100*/;      
   Matrix3d IMUData::_accMeasCov = Matrix3d::Identity()*2.0e-3*2.0e-3/0.005*100; 
   Why did you multiply 100 at the last ? 

2. in function void EdgeNavStatePVRPointXYZOnlyPose::linearizeOplus()

   Vector3d Paux = Rcb*Rwb.transpose()*(Pw-Pwb);
   Matrix<double,2,3> JdRwb = - Jpi * (Sophus::SO3::hat(Paux) * Rcb);
   There's transpose on Rwb, how to get the Jacobian W.R.T.  Rwb ???
   Could you please give me some references ???

Thank you so much !!!

leavesnight, 2017-11-14:
1. I think maybe Jing Wang find accelerate covariance 10 times making the result better. You can try other parameter and compare the RMSE with the groundtruth.
2. I think he uses a right(for Rwb, left for Rbw) disturbance model: Rwb->Rwb*Exp(dphi_wb) or Rbw->Exp(-dphi_wb)*Rbw
I suppose u know the chain rule: JdRwb=J_error_Pc*J_Pc_dRwb=-Jpi* J_Pc_dRwb
and I prefer twb to express the Pwb:
Pc=Rcb*Rbw*(Pw-twb)+tcb, notice tcb is constant
so J_Pc_dRwb=partial(Rcb*Rbw*(Pw-twb))/partial(phi_wb), here phi_wb=Log(Rwb)
=partial(Rcb*Exp(-phi_wb)*(Pw-twb))/partial(phi_wb), here I use Exp(phi_bw)=Exp(-phi_wb)
left(for Rbw) disturbance model=[Rcb*Exp(-dphi_wb)*Exp(-phi_wb)*(Pw-twb)-Rcb*Exp(-phi_wb)*(Pw-twb)]/dphi_wb=Rcb*(-dphi_wb^)*Exp(-phi_wb)*(Pw-twb)/dphi_wb, here use Exp(dphi)=I+dphi^
=Rcb*[Exp(-phi_wb)*(Pw-twb)]^dphi_wb/dphi_wb, here use cross rule:aXb=-bXa or a^b=-b^a
=Rcb*[Rbw*(Pw-twb)]^=[Rcb*Rbw*(Pw-twb)]^Rcb. here use Adjoint property:Rexp(p^)=exp((Rp)^)R
prove over.

einspyon, 2017-11-20:
@leavesnight  
    That's great !!!  
    It's very kind of you to offer me so detailed derivation , which helped me greatly to understand calculus on lie group !!!  
    Thanks very much !!!