jingpang/LearnVIORB #36: make -j error

cumtchenchang, 2017-11-07:
I checked and reinstalled eigen, but also encountered this error.I tryed eigen 3.3 and 3.4.
cc@cc-B250-HD3:~/catkin_ws/src/LearnVIORB-RT/build$ make -j
[  3%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /home/cc/catkin_ws/src/LearnVIORB-RT/src/IMU/imudata.h:4,
                 from /home/cc/catkin_ws/src/LearnVIORB-RT/include/Frame.h:35,
                 from /home/cc/catkin_ws/src/LearnVIORB-RT/include/KeyFrame.h:29,
                 from /home/cc/catkin_ws/src/LearnVIORB-RT/include/MapPoint.h:24,
                 from /home/cc/catkin_ws/src/LearnVIORB-RT/include/Map.h:24,
                 from /home/cc/catkin_ws/src/LearnVIORB-RT/include/Optimizer.h:24,
                 from /home/cc/catkin_ws/src/LearnVIORB-RT/src/Optimizer.cc:21:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:712:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:693:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:682:32:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Derived = Eigen::Matrix<int, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:225:24:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; _Scalar = int; int _Rows = -1; int _Cols = 1; int _Options = 0; int _MaxRows = -1; int _MaxCols = 1]’
/usr/include/eigen3/Eigen/src/Core/PermutationMatrix.h:367:17:   required from ‘Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>& Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>::operator=(const Eigen::PermutationBase<OtherDerived>&) [with Other = Eigen::PermutationMatrix<-1, -1, long int>; int SizeAtCompileTime = -1; int MaxSizeAtCompileTime = -1; _StorageIndex = int]’
/home/cc/catkin_ws/src/LearnVIORB-RT/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:68:18:   required from ‘void g2o::LinearSolverEigen<MatrixType>::CholeskyDecomposition::analyzePatternWithPermutation(g2o::LinearSolverEigen<MatrixType>::SparseMatrix&, const PermutationMatrix&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>; g2o::LinearSolverEigen<MatrixType>::SparseMatrix = Eigen::SparseMatrix<double, 0>; g2o::LinearSolverEigen<MatrixType>::PermutationMatrix = Eigen::PermutationMatrix<-1, -1, long int>]’
/home/cc/catkin_ws/src/LearnVIORB-RT/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:195:9:   required from ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/cc/catkin_ws/src/LearnVIORB-RT/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:100:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, double*, double*) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/cc/catkin_ws/src/LearnVIORB-RT/src/Optimizer.cc:4568:1:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:707:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
   EIGEN_STATIC_ASSERT((internal::functor_is_product_like<BINOP>::ret \
   ^
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:745:3: note: in expansion of macro ‘EIGEN_CHECK_BINARY_COMPATIBILIY’
   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
   ^
CMakeFiles/ORB_SLAM2.dir/build.make:350: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

KunWangV, 2018-04-03:
@cumtchenchang solve it ? I have same problem.

liming0791, 2018-06-08:
same problem

huhumeng, 2018-06-08:
This a simple problem, you two should just google it.

huhumeng, 2018-06-08:
This will occur because of your eigen version, just change the thirdparty/g2o/g2o/solvers/linear_solver_eigen.h line 
// typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, SparseMatrix::Index> PermutationMatrix;
to
typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, int> PermutationMatrix;
why not you google or baidu it first before complainant here.