jingpang/LearnVIORB #37: I fund a bug when I use my device with slow and small vision change motion at the beginnig

wangqi123412431, 2017-11-14:
Hi, 
When I run my own device with a slow movement at the beginning, the VIORB always died. And I finally found the problem and solve it in the code.

sorry, for convenience, Iet me explain it in chinese :)

在LocalMapping线程里，函数AddToLocalWindow(KeyFrame* pKF)在往mlLocalKeyFrames里添加新的关键帧时候，有种情况没有考虑到，会导致系统直接报错退出。就是在系统刚运行的时候，由于相机移动缓慢且视觉变化不大，在LocalMapping线程KeyFrameCulling()中会剔除冗余的关键帧，那么伴随着函数DeleteBadInLocalWindow()，mlLocalKeyFrames里的关键帧也会被剔除，这样会有很大的几率导致mlLocalKeyFrames里面的关键帧从一个接近窗口大小或等于窗口大小的数量在一次关键帧剔除后减少到只剩几帧，于是在新的一次AddToLocalWindow(KeyFrame* pKF)时，会从LocalMapping.cc的864行开始执行，也就是循环地向mlLocalKeyFrames还剩的关键帧的最前的关键帧那帧的前面，插入它前一帧关键帧(有点绕...)，直到数量超过窗口阈值或者是前一帧是空的(NULL)，注意，这个条件是允许把系统最最开始的第一帧插进去的，在这种情况下，mlLocalKeyFrames的最前一帧关键帧是整个系统的第一帧关键帧。那么，问题来了，在局部优化的时候，是需要取mlLocalKeyFrames最前一帧关键帧的前一关键帧作为顶点添加IMU相关约束进行优化的，如果出现以上情况，mlLocalKeyFrames最前一帧关键帧的前一帧是空的，那么就会出问题。具体体现在Optimizer.cc的247行或者是1094行开始向g2o里添加顶点时，系统直接就退出了。

说了这么多，就是在描述一种会出现错误的情况，我的解决办法很简单，就是在AddToLocalWindow(KeyFrame* pKF)函数最后几行里增加一个条件，避免把系统最开始的那帧关键帧也插入mlLocalKeyFrames中去。
` while(mlLocalKeyFrames.size() < mnLocalWindowSize && pKF0->GetPrevKeyFrame()!=NULL)
        {
            pKF0 = pKF0->GetPrevKeyFrame();
            if(pKF0->mnId != 0) //新增加的语句
                mlLocalKeyFrames.push_front(pKF0);
        }`

最后还想说一下，因为自己用的设备是廉价的IMU，实际跑了一下VIORB，发现会出现尺度漂移的现象，所以闭环检测时，计算的还是相似变换，这样就能闭上了，否则几何校验这块过不去，闭环没法闭...

skylook, 2017-11-18:
https://github.com/jingpang/LearnVIORB/issues/33
Is this the same?

imrasp, 2017-11-20:
I have some problem seem similar to you guys VIORB died after LOST tracking ("Track lost soon after initialisation, reseting... ") then System Reseting and nothing happened after that. Did u guys have this problem?.

nanxixiansheng, 2018-04-17:
hello,i want to know how do you run VIORB with your own device? Can you give me a tutorial? Thank you in advance.

CQnancy, 2023-08-04:
> 

Hello, did you solve it later?