jingpang/LearnVIORB #41: A problem about the update of Map scale after running TryInitVIO()

Markbess, 2018-01-18:
Hi.
Here in the functionTryInitVIO(), the system had uptaded the map scales including all mappoints and keyframs, but in the next program, the map scales would be updated again.
What's the difference between these two updates?
here are some code segments in TryInitVIO().
-------------------------------------------------------------------------------------------------------------------------------------
 vector<KeyFrame*> mspKeyFrames = mpMap->GetAllKeyFrames();
            for(std::vector<KeyFrame*>::iterator sit=mspKeyFrames.begin(), send=mspKeyFrames.end(); sit!=send; sit++)
            {
                KeyFrame* pKF = *sit;
                cv::Mat Tcw = pKF->GetPose();
                cv::Mat tcw = Tcw.rowRange(0,3).col(3)*scale;
                tcw.copyTo(Tcw.rowRange(0,3).col(3));
                pKF->SetPose(Tcw);
            }
            vector<MapPoint*> mspMapPoints = mpMap->GetAllMapPoints();
            for(std::vector<MapPoint*>::iterator sit=mspMapPoints.begin(), send=mspMapPoints.end(); sit!=send; sit++)
            {
                MapPoint* pMP = *sit;
                //pMP->SetWorldPos(pMP->GetWorldPos()*scale);
                pMP->UpdateScale(scale);
            }
            std::cout<<std::endl<<"... Map scale updated ..."<<std::endl<<std::endl;
------------------------------------------------------------------------------------------------------------------------
the next code sengments are 
                bool tmpbool = TryInitVIO();
                        SetVINSInited(tmpbool);
                        if(tmpbool)
                        {
                            // Update map scale
                           mpMap->UpdateScale(mnVINSInitScale);
                            // Set initialization flag
                            SetFirstVINSInited(true);
                        }
Thank you in advance.

highlightz, 2018-01-19:
There are two different conditions. The `TryInitVIO` in VINSInitThread is for real-time running, your above-mentioned snippets are for non-realtime running.    
Pay attention to `ConfigParam::GetRealTimeFlag()`, you'll understand this.

Markbess, 2018-01-24:
Actually, I would like to express that the map scale is updated when running TryInitVIO(), and then mpMap->UpdateScale(mnVINSInitScale) would update that again.

closed 2018-01-26 by Markbess

ns15417, 2019-10-31:
@Markbess have you solved this question cuz you closed it. I am also confused why the map is updated twice cuz if it is set to non-realtime mode, the update will be executed both inside TryInitVIO() and LocalMapping::Run(). This leads to some keyframe drifts away when the second update is finished while it is totally ok if i note the second update.
How did you solve this problem ?