jingpang/LearnVIORB #43: Unable to build the RT branch

dharma-kc, 2018-03-08:
OS: Ubuntu 16.04
ROS: kinetic
Eigen: 3.1.0
OpenCV: 3.3.1 [ From /opt/ros/kinetic ] 

Here is the screen shot of the error.

[image]

sunchaothu, 2018-03-18:
I don't know why but below might work.
open Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h，change
```
typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic,SparseMatrix::Index> PermutationMatrix;
```
to
```
typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic,int> PermutationMatrix;
```
I also refered to this link（http://blog.csdn.net/wys422/article/details/78416654）

mrteera, 2018-05-20:
You can also check the build step in my Dockerfile: https://github.com/mrteera/LearnVIORB/blob/RT/Dockerfile

connorsoohoo, 2018-07-30:
For further reference look at [https://github.com/raulmur/ORB_SLAM2/issues/68#issuecomment-220739225](url)