jingpang/LearnVIORB #44: file open error in TryInitVIO

Anthony-chenyb, 2018-03-09:
I meet some problems like "file open error in TryInitVIO", is there any problem with my test.InitVIOTmpPath: ? 

   static bool fopened = false;
    static ofstream fgw,fscale,fbiasa,fcondnum,ftime,fbiasg;
    string tmpfilepath = ConfigParam::getTmpFilePath();
    if(!fopened)
    {
        // Need to modify this to correct path
        fgw.open(tmpfilepath+"gw.txt");
        fscale.open(tmpfilepath+"scale.txt");
        fbiasa.open(tmpfilepath+"biasa.txt");
        fcondnum.open(tmpfilepath+"condnum.txt");
        ftime.open(tmpfilepath+"computetime.txt");
        fbiasg.open(tmpfilepath+"biasg.txt");

i  can't  find these .txt files in my test.InitVIOTmpPath

highlightz, 2018-03-16:
Yes, you should set the path in your config file. For example,
test.InitVIOTmpPath: "/home/your_machine/tmp/"

imrasp, 2018-05-22:
and the folder should be exist.