jingpang/LearnVIORB #45: How to interpret the scale.txt file

dharma-kc, 2018-03-12:
I got the scale.txt file as follows:

1403636580.963556 5.404152 5.401110 
1403636581.113555 5.342806 5.317275 
1403636581.263556 5.383121 5.398332 
1403636581.413555 5.439227 5.319515 
1403636581.563555 5.535811 5.424372 
...
...
....
similar other values
....

I think the first is the timestamp for the keyframe, and what are the two values after that. I found that the first value is s_ and second value is sstar from code, but unable to understand the meaning and difference between them. And does the scale change with each keyframe, or is it constant after we initialize the scale from certain keyframes. Any help will be appreciated. Thank you.

highlightz, 2018-03-16:
`cout<<"Time: "<<mpCurrentKeyFrame->mTimeStamp - mnStartTime<<", sstar: "<<sstar<<", s: "<<s_<<endl;`

I think sstar stands for initially computed scale without considering gravity magnitude as a prior; instead, s_ stands for optimized scale where gravity magnitude serves as a prior constaint.

dharma-kc, 2018-03-22:
@highlightz 
Thank you for your reply. And I got total of 574 scale values, so does it mean that the scale is different for different keyframes ? If that's the case how do I apply this scale to the final map. I mean should I iterate with timestamp and multiply the translation_x, translation_y, and translation_z with the corresponding scale value for the final map or is the scale a single value that I can multiply to the whole map. 
Regards,
Dharma KC

imrasp, 2018-05-22:
As I understand so far, yes it is, scale is updated for every keyframes before map update, and use only last one to update the map(update only one time in whole process). Guess that you fix VINSInitTime value to be infinity number like 99999999 to force process not to update the map for some reason. To apply the scale to final map you can try calling section under `if(bVIOInited)` in function `TryInitVIO` in `localmapping.cc` at the end.