jingpang/LearnVIORB #48: In the RT version, the JdPwb express different from the master version?

jyzhong, 2018-04-17:
@jingpang  in the RT version ,
Matrix<double,3,3> JdPwb = - Jpi * (-Rcb*Rwb.transpose());
while, int the master version ,
Matrix<double,3,3> JdPwb = - Jpi * (-Rcb);

when i try change the expression in the master as the same with  RT version ,i test mh_01_easy with a tracking lost.
what is the correct expression of  JdPwb ?
                                                                                   Hope get your message,thanks!

lianbin, 2018-10-12:
@jyzhong  Have you figured out this problem?I also have this question.

jyzhong, 2018-10-12:
> @jyzhong Have you figured out this problem?I also have this 

> @jyzhong Have you figured out this problem?I also have this question.

Obviously, the right expression is uniqueness. I just wonder why the wrong format is ok.