jingpang/LearnVIORB #64: lost tracking after initialization 15 sec?

amranov, 2018-12-21:
Hi , first i would like to thank you for this great job 
i tried to run the code USING loirorVIsensor but the tracking is lost after initialization , no new KF 
Also the scale value are bit weird (negative values)
IMU and CAMERA i'm using are well synchronized , any succession to solve this issue ?

CQnancy, 2023-08-08:
Have you resolved the issue?

CQnancy, 2023-08-08:
> 
Have you resolved the issue?

CQnancy, 2023-08-08:
> Hi , first i would like to thank you for this great job i tried to run the code USING loirorVIsensor but the tracking is lost after initialization , no new KF Also the scale value are bit weird (negative values) IMU and CAMERA i'm using are well synchronized , any succession to solve this issue ? Time: 15.2993, sstar: 0.710862, s: 0.495204 Local Mapping STOP -----------here is the last KF in vScaleGravityKF------------ dt = 0.000, this vs next time: 1545394068.362 vs 1545394068.362
> 
> ... Map scale updated ...
> 
> ... Map NavState updated ...
> 
> Local Mapping RELEASE
> 
> dt = -0.038, this frame vs last imu time: 1545394085.081 vs 1545394085.043 dt = -0.038, this frame vs last imu time: 1545394085.081 vs 1545394085.043 CHOLMOD warning: L not positive definite

Have you resolved the issue?