jingpang/LearnVIORB #66: Worst result than ORB_SLAM2

giwrgosg, 2019-05-28:
I run ORB_SLAM2 successfully in some datasets, however when I use the exactly same parameters for the visual part the track is lost in two sequences that I tried. In ORB_SLAM2 the tracking was achieved successfully.

Any suggestions?

lidongjiangBJTU, 2019-06-05:
Have you solved that problem? I encountered the same question when i try to run it on D435I dataset.

giwrgosg, 2019-06-07:
No still getting worse results

Shashika007, 2019-06-10:
@lidongjiangBJTU  try to update your D435i Tbc parameter from kalibr and imu_utils. @giwrgosg Try another repo which has less bug fixed.

giwrgosg, 2019-06-10:
@Shashika007 Do you have a suggestion for a specific repo? Because I checked some and none is claiming that has fixed bugs relevant to the actual performance

lidongjiangBJTU, 2019-06-11:
@Shashika007 I have already update the Tbc and imu hz, but i can not get good results

arindamsaha81, 2020-05-18:
@lidongjiangBJTU  @Shashika007  @giwrgosg  Did u get any better result with D435i? Can you please provide me a better Tbs that produces better result for D435i?

CQnancy, 2023-08-04:
> 

Is Tbc Tic in kalibr?