jingpang/LearnVIORB #70: Function optimizer.computeMarginals()

Andrea485, 2019-10-04:
hi, the result of the optimizer.computeMarginals(), is the covariance matrix about the estimated pose?
https://github.com/jingpang/LearnVIORB/blob/b9e73cefee88d7ab352508e1e11aab0990195e91/src/Optimizer.cc#L3745

ferreram, 2019-10-04:
Hi, the function computeMarginals() returns the Hessian matrix associated to the state vertex(0), you have to invert it to get the covariance.

Andrea485, 2019-10-04:
1)thanks for the reply, I tried writing this code:
`g2o::SparseBlockMatrixXd spinv;
    optimizer.computeMarginals(spinv, optimizer.vertex(0));
    Eigen::Matrix<double, 6, 6> margCovInv = Eigen::Matrix<double, 6, 6>::Zero();
    margCovInv.topLeftCorner(6,6) = spinv.block(0, 0)->inverse();
    cout<<"Matrix covariance:\n"<<margCovInv<<endl;`

But the matrix seems to be incorrect (values too high?)

[image]

Did I miss something in the code?

2)This covariance matrix refer only to the last post, or to the full trajectory?

ferreram, 2019-10-04:
1) Well, the values of the Hessian matrix come from this expression: J' W
J, where J is the Jacobian of your optimization problem and W the
information matrix (the inverse of the covariances given to each
measurements).

The resulting Hessian matrix hence depends on the optimization problem you
have defined.  The fact that you get high values for the state vertex(0)
means that you have a high confidence in the estimated parameters (if you
invert this Hessian matrix you will get the covariance matrix associated to
the estimated state).

2) I think that in g2o, the computeMarginals() function simply provides the
block in the Hessian matrix associated to the state you require (but I
don't remember well, you should check that).  If it is indeed done like
this, then the matrix that you get correspond to the confidence you have in
vertex(0) according to the optimization problem you have defined.  Hence,
if you have performed a Global Bundle Adjustment like optimization, the
matrix you have obtained correspond to the information matrix related to
the states vertex(0) over the full trajectory.

arianna003, 2019-10-04:
hi @ferreram , the matrix @Andrea485's figure represent covariance not hessian,  but uncertainty in the pose coordinate in meters?

ferreram, 2019-10-04:
Hi @arianna003 , yes you're right, my bad I read it too fast.  In this case @Andrea485 , the uncertainty values of your covariance matrix indeed seems very high.  The uncertainty around the translation parts of the matrix are in squared meters if the transnational parts of the states are expressed in meters (it should be the case with a Visual-Inertial method but not with a monocular SLAM)

Andrea485, 2019-10-04:
@ferreram  what's wrong in my code, what can I do?

ferreram, 2019-10-04:
@Andrea485 , there is nothing wrong in the code you have used to compute the covariance matrix.  But what are you doing?  VI-Odometry?  If so, is the Bundle Adjustment like optimization is giving you good results?  You can also have a look at the information you are setting for your edges.

As the Hessian matrix is derived from the Jacobian and the information matrix, high covariances will come from high Jacobians value.  Maybe you should verify your Jacobian as well (you can test doing automatic differentiation).