jingpang/LearnVIORB #74: One questions on PoseOptimization

wds01060202, 2020-04-16:
Thank you very much for sharing the code!
I am learning your code recently, and I have one questions on PoseOptimization functions, could you give me some instructions? Thanks in advance!

how you determine the values for Huber?
I see that in PoseOptimization() fuction the parameters is normal.
for example "const float thHuberNavStatePVR = sqrt(21.666);"

ferreram, 2020-04-16:
Hi @wds01060202 ,
The Huber values are taken from the chi2 table, using the number of degrees of freedom of the measurements and an inlier probability (here the 21.66 value was chosen with 9 degrees of freedom as from the 9D vector representing position, rotation, velocity, and with an inlier probability of 1%).
However, for me, it does not make sense to use a robust cost function on the IMU measurements as they indeed follow a gaussian distribution.  The use of Huber is more suited for handling visual measurements as some of them might come from outliers which completely break the gaussian distribution assumption and hence would result in very bad optimization if not discarded by the use of a robust cost function.
My guess for the weird 100 factor used in GlobalBundleAdjustmentNavStatePRV() is that bad results were obtained when keeping the 21.666 value and that better results could be achieved when adding the 100 factor.  However, multiplying by 100 most likely removes the effect of the Huber cost as it is unlikely that the IMU measurements error gets over the Huber threshold.

wds01060202, 2020-04-27:
> Hi @wds01060202 ,
> The Huber values are taken from the chi2 table, using the number of degrees of freedom of the measurements and an inlier probability (here the 21.66 value was chosen with 9 degrees [...]
Hi @ferreram 
Thank you for your reply, I agree your point, i cheak the information Matrix of imu graph edge,and found  it is very huge compared to the graph edge of vision, and the measurements error can gets over the Huber  threshold easily. I guess that the author @jingpang trust the imu more,so he used the weird 100 factor.