jingpang/LearnVIORB #76: How do you marginalize key frames discarded in LocalBAPRVIDP ()

NCHUPJ, 2020-05-07:
How do you marginalize key frames discarded in LocalBAPRVIDP ()

NCHUPJ, 2020-05-07:
When IMU is tightly coupled and optimized, there is no marginalization in the code. won't this cause a lot of information to be lost?

ferreram, 2020-05-07:
This follows the VI-ORB-SLAM paper. Marginalization is only applied when optimizing the current frame (which is linked to the previous keyframe through the preintegrated IMU measurement and constrained by the current marginalization prior).
No information is really lost because a window of previous keyframes is included as fixed constraints in the optimization.  However, compared to a sliding window / marginalization approach such as VINS-Mono, the uncertainty information on the fixed keyframes is lost which might lead to overconfidence during optimization but also ensure faster convergence and small drift.

NCHUPJ, 2020-05-07:
Thank you very much, look at the source code found marginalized only in TrackLocalMapWithIMU();
I can understand that the front has been marginalized, so the sliding window need not be marginalized, because the information of the frames in the sliding window is complete.

NCHUPJ, 2020-05-07:
@ferreram Thank you very much, look at the source code found marginalized only in TrackLocalMapWithIMU();
I can understand that the front has been marginalized, so the sliding window need not be marginalized, because the information of the frames in the sliding window is complete.